platformio/platformio-vscode-ide #358: Start PIO Home Server

klvnraju, 2018-10-04:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.27.2
PIO IDE: v1.3.1
System: Windows_NT, 10.0.17758, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\nkotchar\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\nkotchar\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\nkotchar\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\nkotchar\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 7, in <module>
    from socket import error as SocketError, timeout as SocketTimeout
  File "C:\Users\nkotchar\.platformio\python27\Lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: The specified procedure could not be found.

```

ivankravets, 2018-10-09:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/176

closed 2018-10-09 by ivankravets

ivankravets, 2018-10-24:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/379#issuecomment-432766866?