platformio/platformio-vscode-ide #361: Start PIO Home Server

Sl1nd0, 2018-10-09:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.0
PIO IDE: v1.3.1
System: Windows_NT, 10.0.16299, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\silindokuhle\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\silindokuhle\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\silindokuhle\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\silindokuhle\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 7, in <module>
    from socket import error as SocketError, timeout as SocketTimeout
  File "C:\Users\Silindokuhle\.platformio\python27\Lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: The specified procedure could not be found.

```

ivankravets, 2018-10-09:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/176

closed 2018-10-09 by ivankravets

ivankravets, 2018-10-24:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/379#issuecomment-432766866?