platformio/platformio-vscode-ide #363: Start PIO Home Server

zekahh, 2018-10-09:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.0
PIO IDE: v1.4.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\zekah\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\zekah\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\zekah\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\zekah\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 7, in <module>
    from socket import error as SocketError, timeout as SocketTimeout
  File "C:\Users\Zekah\.platformio\python27\Lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: The specified procedure could not be found.

```

ivankravets, 2018-10-10:
Is this issue related to https://github.com/platformio/platformio-vscode-ide/issues/176 ?

zekahh, 2018-10-10:
Hey Ivan!, thanks for your answer!, you're working a lot,

Well this didn't solve the problem "I was trying to make platformio work
with atom, and I got this image
[image: image.png]

then I tried with Vscode, xD and somthing similar happened

[image: image.png]
actually, is still loading, so, I'm trying to fix this  uninstalling all
the python versions from my laptop, and others tricks...

Cheers!

ivankravets, 2018-10-10:
Hey! ;) So, does it work now?

zekahh, 2018-10-11:
Hey Ivan!,

Hey Hey!, I was writting the e-mail, when  PIO wasn't working yet,  Now it
works! :D

https://community.platformio.org/t/home-not-loading/2579 , I did what
CrazyMerlin did, way exaggerated tho xD

how did I fix it?,  I did, (a lot of stuff and didn't work), until ..
I  deleted all the python versions manually,
 then uninstalled platformio from Atom,
then I deleted Atom,
then I deleted all the .platformio folders in* C:/users/nick, *
then deleted the .atom folder.
reset my laptop, installed atom again, then downloaded the pio-ide package,
and now it works :D!

by the way, now I understand a little bit why this happened reading a post
about multiple PIOs in your system, and
Yesterday I installed PIO in Atom and it worked very well, then I wanted to
make it run in VScode, so, it seems I created two PIOs inside my laptop,
and somehow it burnt  all, creating issues for free e.e.

[image: image.png]

and well that's it C:

Cheers Ivan, have a great day/night !!!

ivankravets, 2018-10-12:
How did you resolve it?

closed 2018-10-12 by ivankravets