platformio/platformio-vscode-ide #366: Start PIO Home Server

miguelpom, 2018-10-11:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.27.1
PIO IDE: v1.4.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\casfid\.platformio\packages\contrib-pysite\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\casfid\.platformio\packages\contrib-pysite\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 no es una aplicaci�n Win32 v�lida.

```

miguelpom, 2018-10-11:
OK!
I should learn the lesson: **SEARCH FOR DUPLICATES BEFORE REPORTING AN ISSUE!!!**

closed 2018-10-11 by miguelpom

ivankravets, 2018-10-11:
:) 
How did you resolve it?