platformio/platformio-vscode-ide #367: Start PIO Home ServerPl, Any PIO command works only in sudo

Kameeno, 2018-10-12:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.1
PIO IDE: v1.4.0
System: Darwin, 17.0.0, x64

# Exception
```
Error: Traceback (most recent call last):
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/commands/home.py", line 32, in cli
    pioplus_call(sys.argv[1:])
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/core.py", line 106, in pioplus_call
    pioplus_path = join(get_core_package_dir("tool-pioplus"), "pioplus")
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/core.py", line 82, in get_core_package_dir
    pkg_dir = pm.get_package_dir(name, requirements)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 381, in get_package_dir
    manifest = self.get_package(name, requirements, url)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 349, in get_package
    for manifest in self.get_installed():
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 339, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 301, in load_manifest
    src_manifest_path = self.get_src_manifest_path(pkg_dir)
  File "/Users/kameeno/.platformio/penv/lib/python2.7/site-packages/platformio/managers/package.py", line 272, in get_src_manifest_path
    for item in os.listdir(pkg_dir):
OSError: [Errno 13] Permission denied: '/Users/kameeno/.platformio/packages/framework-arduinoteensy'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

Kameeno, 2018-10-12:
on `pio run` 

`IOError: [Errno 13] Permission denied: '/usr/local/Cellar/platformio/3.6.0/libexec/lib/python2.7/site-packages/idna-2.7-py2.7.egg-info/PKG-INFO'`

ivankravets, 2018-10-12:
It seems that you install `homebrew` or `pio` as root. It's so dangerous. 

1. Please uninstall PIO as root and install it without root.
2. Remove this folder with root permissions => /Users/kameeno/.platformio

closed 2018-10-12 by ivankravets