platformio/platformio-vscode-ide #368: Cannot compile when "WeMos WiFi & Bluetooth Battery (WEMOS)" board is selected

trophysoft, 2018-10-12:
When trying to compile even the simplest "hello world" project for the "WeMos WiFi & Bluetooth Battery (WEMOS)" board (board=wemosbat in platformio.ini), the following errors are shown:

Generating partitions .pioenvs\wemosbat\partitions.bin
Archiving .pioenvs\wemosbat\libFrameworkArduinoVariant.a
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
'Bluetooth' is not recognized as an internal or external command,
operable program or batch file.
*** [.pioenvs\wemosbat\src\main.cpp.o] Error 1
Compiling .pioenvs\wemosbat\FrameworkArduino\Esp.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
'Bluetooth' is not recognized as an internal or external command,
operable program or batch file.
*** [.pioenvs\wemosbat\FrameworkArduino\Esp.cpp.o] Error 1

When selecting another board, e.g. nodemcu-32s, everything works fine, but the board that I use is unfortunately the board that I have is the "WeMos WiFi & Bluetooth Battery (WEMOS)" board, and I want to be able to use all of its features.

ivankravets, 2018-10-12:
Please share your `platformio.ini`.

Herschdorfer, 2018-12-30:
I just created a new project and ran into the same issue. 

platformio.ini
`
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:wemosbat]
platform = espressif32
board = wemosbat
framework = arduino
`

Herschdorfer, 2018-12-30:
I have found the issue, the problem is the board name in wemosbat.json. The '&' destroys the build.