platformio/platformio-vscode-ide #371: Start PIO Home Server

schaelsick88, 2018-10-17:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.1
PIO IDE: v1.4.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\Patrick\.platformio\packages\contrib-pysite\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\Patrick\.platformio\packages\contrib-pysite\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 ist keine zul�ssige Win32-Anwendung.

```

ivankravets, 2018-10-23:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/176

closed 2018-10-23 by ivankravets

ivankravets, 2018-10-24:
Could you try this solution https://github.com/platformio/platformio-vscode-ide/issues/379#issuecomment-432766866?