platformio/platformio-vscode-ide #374: Start PIO Home Server

shaukatali10, 2018-10-18:
# trouble running platformio on mac
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.2
PIO IDE: v1.4.1
System: Darwin, 16.0.0, x64

# Exception
```
Error: Detected unknown package 'tool-pioplus'

```

ivankravets, 2018-10-23:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

shaukatali10, 2018-11-03:
Did you uninstall it from Visual Studio or Terminal

ivankravets, 2018-11-03:
Do you use a proxy? Firewall?

Please open PIO IDE Terminal and type
```
pio home
```

Do you see any errors?

sclarke81, 2018-11-07:
Please see [this comment](https://github.com/platformio/platformio-vscode-ide/issues/96#issuecomment-436549492).

shaukatali10, 2018-11-07:
Ive got plenty of errors with regards to the console as follows:
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] Error: Error: Detected unknown package 'tool-pioplus'

	at /Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:14:11760
	at ChildProcess.y (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at maybeClose (internal/child_process.js:925:16)
	at Socket.stream.socket.on (internal/child_process.js:346:11)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at Pipe._handle.close [as _onclose] (net.js:554:12)
t.log @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] runCommand platformio Array(7) Object
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] Error: Error: Detected unknown package 'tool-pioplus'

	at /Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:14:11760
	at ChildProcess.y (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at maybeClose (internal/child_process.js:925:16)
	at Socket.stream.socket.on (internal/child_process.js:346:11)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at Pipe._handle.close [as _onclose] (net.js:554:12)
t.log @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] runCommand platformio Array(7) Object
2/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] Error: Error: Detected unknown package 'tool-pioplus'

	at /Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:14:11760
	at ChildProcess.y (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at maybeClose (internal/child_process.js:925:16)
	at Socket.stream.socket.on (internal/child_process.js:346:11)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at Pipe._handle.close [as _onclose] (net.js:554:12)
t.log @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] runCommand platformio Array(7) Object
2/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282 [Extension Host] rejected promise not handled within 1 second
t.log @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:282
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297 [platformio.platformio-ide]HTTP Error (429): Creation of this event was denied due to rate limiting
e.$onExtensionRuntimeError @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297 e: HTTP Error (429): Creation of this event was denied due to rate limiting
	at new e (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:22094)
	at ClientRequest.<anonymous> (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:27336)
	at Object.onceWrapper (events.js:315:30)
	at emitOne (events.js:116:13)
	at ClientRequest.emit (events.js:211:7)
	at HTTPParser.parserOnIncomingClient (_http_client.js:551:21)
	at HTTPParser.parserOnHeadersComplete (_http_common.js:117:23)
	at TLSSocket.socketOnData (_http_client.js:440:20)
	at emitOne (events.js:116:13)
	at TLSSocket.emit (events.js:211:7)
	at addChunk (_stream_readable.js:263:12)
	at readableAddChunk (_stream_readable.js:250:11)
	at TLSSocket.Readable.push (_stream_readable.js:208:10)
	at TLSWrap.onread (net.js:594:20)
e.$onExtensionRuntimeError @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297 [platformio.platformio-ide]HTTP Error (429): Creation of this event was denied due to rate limiting
e.$onExtensionRuntimeError @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297
/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.main.js:3297 e: HTTP Error (429): Creation of this event was denied due to rate limiting
	at new e (/Users/shockmax/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:22094)
[... cut ...]
	at maybeClose (internal/child_process.js:925:16)
	at Socket.stream.socket.on (internal/child_process.js:346:11)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at Pipe._handle.close [as _onclose] (net.js:554:12)
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180

ivankravets, 2018-11-19:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/96

closed 2018-11-19 by ivankravets

sclarke81, 2018-11-30:
PIO is installing now, but when it tries to open PIO Home I get this error. I've tried uninstalling the extension, deleting the extension folder (`C:\opt\.vscode\extensions\platformio.platformio-ide-1.5.0`) and deleting `C:\Users\<username>\.platformio`. After reinstalling I still have the same error.

# Configuration

VSCode: 1.29.1
PIO IDE: v1.5.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Error: Detected unknown package 'tool-pioplus'

	at C:\opt\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\platformio-node-helpers\dist\index.js:1:14992
	at ChildProcess.h (C:\opt\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\opt\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```