platformio/platformio-vscode-ide #378: inellisense for tasmota

schroeca, 2018-10-23:
Intellisense does not work with Tasmota SonOff firmware.
I get over 160 errors in which vscode complains about functions which can not be found.

Compiling and uploading works well.

First error appears in sonoff.ino
[image]

The function itself is defined in sonoff_support.ino

Setup:
tasmota release 6.2 
vscode 1.28.2 
platformio: Home 1.0.2·Core 3.6.1rc5

_I'm pretty new to all this stuff, could you please give me an advice on how to fix this?_

ivankravets, 2018-10-23:
INO is not C/C++ format. This is a "crutch" by Arduino IDE. We recommend converting your INO to CPP. See http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

closed 2018-10-23 by ivankravets

schroeca, 2018-10-24:
thx for the clarification.

Changing the intellisense mode to "tag parser" will also remove the errors.
This is no "real" solution but it helps to find real errors.