platformio/platformio-vscode-ide #379: Start PIO Home Server

ww34ww34ww34, 2018-10-23:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.2
PIO IDE: v1.4.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\whitew_fisso\.platformio\packages\contrib-pysite\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\whitew_fisso\.platformio\packages\contrib-pysite\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 non � un'applicazione di Win32 valida.

```

ivankravets, 2018-10-23:
Do you have antivirus or firewall in a system?

denchii, 2018-10-24:
Got the same problem too. Is's started today.

ivankravets, 2018-10-24:
@denchii do you have Windows Firewall enabled? Does `pio account show` command work for you?

denchii, 2018-10-24:
@ivankravets, yes Windows Firewall enabled. Tried to switch it off and add an exception for VSCode, but the problem is still the same. pio account show also didn't work

ivankravets, 2018-10-24:
@denchii could you try to replace `%USERHOME%/.platformio/packages/tool-pioplus/pioplus.exe` with https://www.dropbox.com/s/wnfpx1z602fix0k/pioplus.exe.zip?dl=1

1. Close VSCode
2. Open system task manager and kill all `pioplus` processes
3. Start VSCode again.

Does it work now?

Faynman, 2018-10-25:
@ivankravets  Thanks, it works

ww34ww34ww34, 2018-10-25:
Hi. No no firewall at all. The only way I can make it work was to uninstall all my python release
- aptana 3.4 64 bit
- aptana 3.5 64 bit
- python 2.7 64 bit

and install a 2.7 32 bit.

denchii, 2018-10-25:
@ivankravets yes, it works. Is this the new version that will be released soon?