platformio/platformio-vscode-ide #390: Start PIO Home Server

turnbeutel, 2018-10-28:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.28.2
PIO IDE: v1.4.3
System: Darwin, 18.0.0, x64

# Exception
```
Error: Error: You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.

	at /Users/maskiertemuchachos/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:14:11760
	at ChildProcess.y (/Users/maskiertemuchachos/.vscode/extensions/platformio.platformio-ide-1.4.3/node_modules/platformio-node-helpers/lib/index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ivankravets, 2018-10-30:
Could you reproduce this issue after VSCode restart?

turnbeutel, 2018-10-30:
Hi Ivan, yes after restart I could reproduce the issue. And I found a solution. 
In my case on my machine I've installed an MacOs software called 
Little Snitch to prevent unwanted network traffic. 
The time I do start VSCode, I grant it permissions to bypass Little Snitch.
VSCode cannot/is not able to relay these same permissions the time it starts PIO.

Calling pio from console and granting permissions or disabling of Little Snitch 
will eliminate the problem. 

My bad, but probably somebody else will have the same issue.

closed 2018-10-30 by turnbeutel