platformio/platformio-vscode-ide #401: Anti-Virus vs PlatformIO

andrespineda, 2018-11-06:
I am running VSCode on a Windows10 machine. I installed VSCode and Platformio last week. Every time I open up VSCode, I get the hourglass on the home page. This means I can't select any Boards or Platforms as they also get the hourglass.
The only workaround is to turn off my anti-virus and reboot Windows. This is very annoying, time consuming and potentially dangerous as I have to run without protection.

Am I doing something wrong or can this be fixed? 
See attached screenshot.

[image]

ivankravets, 2018-11-06:
Sorry for the issue. How to reproduce it? What is your anti-virus? What are the settings for anti-virus? It seems that anti-virus block local PIO Home Server (backend).

Please click on Menu > Help > Toggle Developer Tools > Console. Do you see any errors? Please share them here. Thanks!

andrespineda, 2018-11-07:
Thanks for the quick response. I am using Windows10, with Kaspersky
Internet Security product.

1. Reproduction of problem is easy, just open up VSCode and the landing
page for PlatformIO shows spinning hourglass - see screenshots. I can't do
anything else to start or open a project until I close Kaspersky and reboot.
2. Opened up Developer Mode in VSCode and do not see any errors on Console
- see screenshot.
3. Checked other Developer menu items and nothing seems amiss, but
something is blocking the access to the requests from the Core, Recent
News, Recent Projects, Boards, etc panels. See screenshots.
4. I am using Kaspersky and made sure the PIO application was not blocked.
See screenshots of settings where I made sure PIO and related apps were
allowed to communicate.
5. What is the URL:port of the backend server? Maybe if I run it from a
browser, it will provide more details.

Thanks for your help and I will provide whatever you need to troubleshoot
this issue.

best,
Andy
[image: pio_1.JPG]
[image: FW2.JPG]
[image: FW1.JPG]
[image: PlatformErrors.JPG]

andrespineda, 2018-11-07:
Well, after all that, I found a newer version of  Kaspersky and updated the software. After installing a reboot, PlatformIO is working with no issues.
Sorry about all the trouble and thanks for the help. I am closing this issue.

closed 2018-11-07 by andrespineda