platformio/platformio-vscode-ide #402: 'Show at startup' state not saved on linux

hanneskuettner, 2018-11-07:
Similar to #81 I have the problem that the state of the 'Show at startup' checkbox is not saved and thus the home page is shown on every startup.
The solution mentioned in #81 (reinstalling the extension) did not work for me. The issue persists.

My `homestate.json` looks as follows
```
{"storage": {"homeDir": "/home/hannes", "coreCaller": "vscode", "cid": "12410e14-52e4-5021-93bd-a7e6dac5d252", "projectsDir": "/home/hannes/Documents/PlatformIO/Projects", "lastCheckPlatformUpdates": 1541151480587, "coreVersion": "3.6.2a2", "routeBadges": {}, "lastCheckLibraryUpdates": 1541151480587, "recentProjects": ["/home/hannes/projects/mi-casa"], "coreSystype": "linux_x86_64"}}
```

hanneskuettner, 2018-11-18:
I can confirm that this resolves my issue.
Maybe an on change event for updating the `homestate.json` file should be considered.

Thanks anyways!

ivankravets, 2018-11-18:
There is a delay for 5sec for batch operations.

closed 2018-11-18 by ivankravets