platformio/platformio-vscode-ide #405: Start PIO Home Server

Lotharius96, 2018-11-10:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.31.2
PIO IDE: v2.2.4
System: Linux, 4.15.0-20-generic, x64

# Exception
```
Error: Error: spawn platformio ENOENT
    at /home/lotharius16/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:14:11760
    at y (/home/lotharius16/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1:13369)
    at ChildProcess.<anonymous> (/home/lotharius16/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1:13132)
    at emitOne (events.js:116:13)
    at ChildProcess.emit (events.js:211:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:196:12)
    at onErrorNT (internal/child_process.js:372:16)
    at _combinedTickCallback (internal/process/next_tick.js:138:11)
    at process._tickCallback (internal/process/next_tick.js:180:9)
```

ivankravets, 2018-11-18:
Have you resolved this issue?

ivankravets, 2018-11-19:
Please switch to PlatformIO IDE for VSCode => https://platformio.org/platformio-ide

closed 2018-11-19 by ivankravets