platformio/platformio-vscode-ide #407: platformio switches to serial console immediately after build error

Spongman, 2018-11-13:
if you have the serial monitor open when you invoke the 'upload' command, platformio switches back to serial console immediately after any build errors, effectively removing the build error from the screen. if you're not keeping a keen eye on the build output window during the build, you may miss the fact that the build/upload failed, and waste a bunch of time wondering why the changes you just made aren't reflected on the device.

it would be _much_ better if the build stopped after an error and **didn't** switch back to the serial monitor.

ivankravets, 2018-11-18:
There is no API in VSCode to get a status code of the latest command run in terminal. Meanwhile, you can disable automatic serial port monitor opening. See http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-autocloseserialmonitor

Spongman, 2018-11-18:
There must be another way. Maybe:

1) the build writes the success/failure state to some location accessible to the plugin. 
2) or the build itself triggers the opening of the serial console when it succeeds.

sirius0xff, 2019-07-31:
Bump.
It is rather inconvenient for such convenient feature.

Panometric, 2020-08-21:
Bump some more, it does it if the upload fails also. It absolutely should not switch and indicate you are monitoring new code when you are still looking at the old code. 

https://community.platformio.org/t/if-build-fails-monitor-opens-anyway-covering-error/15529/3

Panometric, 2020-08-21:
> There is no API in VSCode to get a status code of the latest command run in terminal. Meanwhile, you can disable automatic serial port monitor opening. See http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-autocloseserialmonitor

Actually, the option is to close the serial monitor, not to open it, so there is no workaround.

ivankravets, 2021-02-02:
Fixed in the latest release.

closed 2021-02-02 by ivankravets