platformio/platformio-vscode-ide #408: Start PIO Home Server

karanjeet1, 2018-11-13:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.32.2
PIO IDE: v2.2.4
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: 'platformio' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn platformio ENOENT
    at C:\Users\Installer\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:14:11760
    at y (C:\Users\Installer\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:13369)
    at ChildProcess.<anonymous> (C:\Users\Installer\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1:13132)
    at emitOne (events.js:116:13)
    at ChildProcess.emit (events.js:211:7)
    at ChildProcess.cp.emit (C:\Users\Installer\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:30:37)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
```

ivankravets, 2018-11-18:
Have you resolved this issue?

karanjeet1, 2018-11-19:
No sir, not yet.
I am new to this software so can you please help me out in resolving this issue.

ivankravets, 2018-11-19:
Please switch to PlatformIO IDE for VSCode => https://platformio.org/platformio-ide

closed 2018-11-19 by ivankravets