platformio/platformio-vscode-ide #411: Custom project task

Razikus, 2018-11-15:
For now there is no possibility to run custom target in GUI.

It will be very nice to have this option somehow automated.

Currently everything is static

According to this topic:
https://community.platformio.org/t/how-to-create-custom-project-task-rename-it/5391

proddy, 2019-07-15:
+1 for this feature. Right now I'm doing it with VSC tasks (https://code.visualstudio.com/docs/editor/tasks#vscode)

ivankravets, 2019-07-23:
See docs http://docs.platformio.org/en/latest/ide/vscode.html#custom-tasks

closed 2019-07-23 by ivankravets

zapta, 2022-04-23:
The custom tasks described in the link appear in Terminal | Run Tasks...  Is it possible to surface them also in the Project Tasks list?  (The one that is opened by the alien icon in VSCode).

Also, would be nice to be able to associate them with the icons at the bottom of the screen. For example a custom Upload task.

ivankravets, 2022-04-23:
@zapta yes, you need [Custom Targets](https://docs.platformio.org/en/latest/scripting/custom_targets.html).

zapta, 2022-04-23:
Thanks Ivan. I defined the custom task below and can run it with 'pio run --target pioenv' but I don't see it in the left side Platformio/Alien menu (PROJECT TASTS and QUICK ACCESS).   Is there a way to add it there?  I presume under the name Core Env.

```
Import("env")

# Multiple actions
env.AddCustomTarget(
    name="pioenv",
    dependencies=None,
    actions=[
        "python --version",
        "pio run --target erase --target upload --target monitor"
    ],
    title="Core Env",
    description="Show PlatformIO Core and Python versions"
)
```

zapta, 2022-04-26:
> Did you press the Refresh button in the task list?

Thanks Ivan, I can see it now.