platformio/platformio-vscode-ide #412: esptool Spaces in Path

nicholas1230, 2018-11-15:
When trying to upload a complied program onto my ESP32 an error occurs stating that " No such file or directory:". I was able to repeat this on multiple computers which all have the latest PIO and VScode installed. The problem seems to be the space between "name surname" in the path. The verbose Upload is given below:

>  Executing task: C:\Users\name surname\.platformio\penv\Scripts\platformio.exe run --verbose --target upload 
> 
> Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
> CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
> PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
> HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
> DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
> Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF MODES: FINDER(chain) COMPATIBILITY(soft)
> Collected 0 compatible libraries
> Scanning dependencies...
> No dependencies
> <lambda>(["checkprogsize"], [".pioenvs\esp32doit-devkit-v1\firmware.elf"])
> MethodWrapper(["checkprogsize"], [".pioenvs\esp32doit-devkit-v1\firmware.elf"])
> Memory Usage -> http://bit.ly/pio-memory-usage
> DATA:    [          ]   3.8% (used 12392 bytes from 327680 bytes)
> PROGRAM: [=         ]  12.3% (used 129047 bytes from 1048576 bytes)
> .pioenvs\esp32doit-devkit-v1\firmware.elf  :
> section            size         addr
> .rtc.text             0   1074528256
> .rtc_noinit           0   1342177280
> .iram0.vectors     1024   1074266112
> .iram0.text       31088   1074267136
> .dram0.data        8208   1073479680
> .noinit               0   1073487888
> .dram0.bss         4184   1073487888
> .flash.rodata     21324   1061158944
> .flash.text       67403   1074593816
> .comment            245            0
> .xtensa.info         56            0
> .debug_line       73725            0
> .debug_info       92203            0
> .debug_abbrev     15409            0
> .debug_aranges     1464            0
> .debug_str         5976            0
> .debug_frame       2296            0
> .debug_loc        76607            0
> .debug_ranges       952            0
> Total            402164
> <lambda>(["upload"], [".pioenvs\esp32doit-devkit-v1\firmware.bin"])
> AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
> CURRENT: upload_protocol = esptool
> MethodWrapper(["upload"], [".pioenvs\esp32doit-devkit-v1\firmware.bin"])
> Auto-detected: COM11
> "c:\users\name surname\.platformio\penv\scripts\python.exe" "C:\Users\name surname\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM11" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 c:\Users\name surname\Documents\PlatformIO\Projects\test\.pioenvs\esp32doit-devkit-v1\bootloader.bin 0x8000 c:\Users\name surname\Documents\PlatformIO\Projects\test\.pioenvs\esp32doit-devkit-v1\partitions.bin 0x10000 .pioenvs\esp32doit-devkit-v1\firmware.bin
> usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
> [--flash_mode {keep,qio,qout,dio,dout}]
> [--flash_size FLASH_SIZE]
> [--spi-connection SPI_CONNECTION] [--no-progress]
> [--verify] [--compress | --no-compress]
> <address> <filename> [<address> <filename> ...]
> esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'c:\\Users\\name'
> *** [upload] Error 2
> ========================== [ERROR] Took 19.31 seconds ==========================
> The terminal process terminated with exit code: 1

nicholas1230, 2018-11-15:
> PS C:\Users\name surename> platformio update
> Updating tool-scons                      @ 2.20501.4      [Up-to-date]
> Updating tool-unity                      @ 1.20403.0      [Up-to-date]
> Updating contrib-pysite                  @ 0.3.2          [Up-to-date]
> Updating contrib-piohome                 @ 1.0.6          [Up-to-date]
> Updating tool-pioplus                    @ 1.5.3          [Up-to-date]
> 
> Platform Manager
> Platform Espressif 32
> Updating espressif32                     @ 1.5.0          [Up-to-date]
> Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]
> Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
> Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
> Updating framework-espidf                @ 3.30101.0      [Up-to-date]
> Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
> 
> 
> Library Manager

ivankravets, 2018-11-18:
Duplicate of https://github.com/platformio/platform-espressif32/issues/118

ivankravets, 2018-11-18:
What is your operating system locale?

closed 2018-11-18 by ivankravets