platformio/platformio-vscode-ide #415: Installation Manager

nimaltd, 2018-11-18:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.3
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C868F0>: Failed to establish a new connection: [Errno 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C86DD0>: Failed to establish a new connection: [Errno 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond',)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C55F90>: Failed to establish a new connection: [Errno 10061] No connection could be made because the target machine actively refused it',)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C55E10>: Failed to establish a new connection: [Errno 10061] No connection could be made because the target machine actively refused it',)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C86950>: Failed to establish a new connection: [Errno 10061] No connection could be made because the target machine actively refused it',)': /platformio/platformio/archive/develop.zip
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio/archive/develop.zip (Caused by NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x03C863F0>: Failed to establish a new connection: [Errno 10061] No connection could be made because the target machine actively refused it',))

You are using pip version 18.0, however version 18.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

 If you have antivirus software in a system, try to disable it for a while.
	at C:\Users\nima\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:14:28389
	at ChildProcess.y (C:\Users\nima\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\nima\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ivankravets, 2018-11-18:
Do you use a proxy?

nimaltd, 2018-11-18:
No.

ivankravets, 2018-11-18:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

nimaltd, 2018-11-19:
C:\Users\nima>echo %path%
C:\Program Files (x86)\STMicroelectronics\st_toolset\asm;C:\Program Files (x86)\Python36-32\Scripts\;C:\Program Files (x86)\Python36-32\;C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\SysGCC\raspberry630\bin;C:\SysGCC\MinGW32\bin;C:\Program Files\PuTTY\;C:\WINDOWS\system32\config\systemprofile\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\SysGCC\raspberry\bin;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Atmel\sam-ba_2.15\drv\;C:\Program Files (x86)\Atmel\sam-ba_2.15;C:\Program Files (x86)\GNU Tools ARM Embedded\7 2017-q4-major\bin;C:\Users\nima\AppData\Local\Microsoft\WindowsApps;C:\SysGCC\arm-eabi.4.9.2\bin;C:\SysGCC\raspberry\bin;C:\SysGCC\arm-eabi\bin;C:\SysGCC\arm-eabi_4_9_2\bin;C:\SysGCC\Raspberry_nosys\bin;C:\Users\nima\AppData\Local\Programs\Microsoft VS Code\bin

C:\Users\nima>echo %pythonpath%
%pythonpath%

C:\Users\nima>python --version
Python 3.6.5

C:\Users\nima>pip --version
pip 10.0.1 from c:\program files (x86)\python36-32\lib\site-packages\pip (python 3.6)

C:\Users\nima>pip list
Package         Version
--------------- -------
asn1crypto      0.24.0
bcrypt          3.1.4
cffi            1.11.5
click           6.7
cryptography    2.2.2
fabric          2.0.0
Flask           1.0.2
Flask-SocketIO  3.0.0
gunicorn        18.0
idna            2.6
invoke          1.0.0
itsdangerous    0.24
Jinja2          2.10
MarkupSafe      1.0
paramiko        2.4.1
Pillow          2.9.0
pip             10.0.1
pyasn1          0.4.2
pycparser       2.18
pygame          1.9.3
PyNaCl          1.2.1
Pypubsub        4.0.0
python-engineio 2.1.0
python-socketio 1.9.0
setuptools      39.0.1
six             1.11.0
webcam-streamer 1.0.5
Werkzeug        0.14.1
wxPython        4.0.3
You are using pip version 10.0.1, however version 18.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

ivankravets, 2018-11-19:
What is your Python version in `C:\Python27`? Please run
```
C:\Python27\python.exe --version
```

nimaltd, 2018-11-19:
I solved problem with disableing antivirus

ivankravets, 2018-11-19:
Thanks for the report! Yes, Antivirus can block PlatformIO Core installer.

Happy coding with PlatformIO!

closed 2018-11-19 by ivankravets

Siddhanta12345, 2023-05-25:
After trying every possible things I am getting this:
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001A96E7DA0B0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001A96E7DA3B0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001A96E7DA560>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001A96E7DA710>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x000001A96E7DA8C0>: Failed to establish a new connection: [Errno 11001] getaddrinfo failed')': /simple/pip/
WARNING: There was an error checking the latest version of pip.