platformio/platformio-vscode-ide #421: Intellisense: Remove spurious "unknown register name 'r0'" errors

conklech, 2018-11-20:
With a default project on the AVR platform (e.g. `megaatmega2560`), calls to some common AVR macros (e.g. `pgm_read_word`) produce spurious errors in the automatic on-save Intellisense, i.e. "unknown register name 'r0'". This is caused by the vscode-cpptools extension calling avr-gcc without the correct command line arguments to specify the AVR MCU in use. See [this not-bug on their issue tracker](https://github.com/Microsoft/vscode-cpptools/issues/2377).

The PlatformIO-generated c_cpp_properties.json needs to be modified to include the correct `-mmcu` option in ""compilerPath". (Which is badly named; per the vscode-cpptools docs, that's where compiler flags should go.) This is board-dependent. For me (board = megaatmega2560), adding  "-mmcu=atmega2560" eliminates the errors—until PlatformIO next overwrites the file.

(There are some other issues with vscode-cpptools intellisense; for example, it doesn't show any warnings on save, although it usually will pick up warnings from a PlatformIO build. Not sure why that is; adding `-Wall` to "compilerPath" does *not* fix that.)

ivankravets, 2018-11-20:
Could you provide a simple code to reproduce this issue?

ivankravets, 2018-11-21:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
```

You should be upgraded to PIO Core 3.6.2b5

Now, please remove `.vscode` folder from a project and restart VSCode. Does it work now?

ivankravets, 2018-11-21:
Resolved in https://github.com/platformio/platformio-core/commit/4a17a9b5b3f6cc4ecbf68b0c0e0d2af5a660e57c

closed 2018-11-21 by ivankravets

conklech, 2018-11-21:
Confirmed that the reported issue is resolved in dev. I didn't need to remove `.vscode` or restart; "Rebuild Intellisense Index" did the right thing. Thanks for the quick fix! Would you still like a test case? Sorry that I missed the earlier request.

There are still problems with intellisense; it reports a markedly different set of warnings/errors than the actual build process. It's not yet clear to me whether it's a vscode-cpptools issue or a platformio issue. I may open a new issue with a good test case.

ivankravets, 2018-11-21:
Yes, please provide a test project to reproduce other issues. Thanks!

fredlcore, 2019-03-17:
@ivankravets: I have upgraded the way you described above, PlatformIO version now is 4.0.0a5, afterwards I removed .vscode folder, but the errors mentioned above still show up, for me more than 70 times
`unknown register name 'r30'`
basically each time I use pgm_read_byte_far() in my code.
Is there any other action I could take?
Thanks!

ivankravets, 2019-03-20:
@fredlcore do you use the latest VSCode and have updated all extensions? What is your `platformio.ini`?

fredlcore, 2019-03-20:
@ivankravets I think everything is up-to-date:
`PlatformIO 4.0.0a5 is currently the newest version available.`
`MacBook-Air-10757:BSB-LAN frederik$ platformio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.181225    [Up-to-date]
Updating contrib-piohome                 @ 2.0.0          [Up-to-date]
Updating tool-pioplus                    @ 2.1.2          [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.12.5         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Updating framework-arduinoavr            @ 2.10623.190209 [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.4          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.190223 [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]

Library Manager
===============
Updating Dusk2Dawn                       @ 1.0.1          [Up-to-date]
Updating ESP8266Ping                     @ 1.0            [Up-to-date]
Updating Ethernet                        @ 2.0.0          [Up-to-date]
Updating Time                            @ 1.5            [Up-to-date]
Updating TinyGPSPlus                     @ 1.0.2          [Up-to-date]`

VSCode version is 1.32.3 and says no newer version available.
I have closed VSCode and removed .vscode folder as recommended, all to no avail.
My platformio.ini is this:
`[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino`

Deoptim, 2020-03-27:
The same problem appeared as soon as I use a driver that, by the way, successfully compiled on Atmel Studio.
PlatformIO latest at the moment, here are the logs:
pio update
Updating contrib-piohome                 @ 3.1.1          [Up-to-date]
Updating contrib-pysite                  @ 2.37.191017    [Up-to-date]
Updating tool-unity                      @ 1.20500.200322 [Up-to-date]
Updating tool-scons                      @ 3.30102.0      [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 2.0.0          [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.190710 [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.4.0          [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]

platformio.ini
[env:attiny13a]
platform = atmelavr
board = attiny13a
board_build.mcu = attiny13a
board_build.f_cpu = 1000000L

upload_protocol = usbasp
upload_flags = -Pusb

[image]

Editing **c_cpp_properties.json** does nothing more so already there is an option

>             "compilerArgs": [
>                 "-mmcu=attiny13a",
>                 ""
>             ]