platformio/platformio-vscode-ide #423: Installation Manager

Antecer, 2018-11-21:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.5
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: PIP: Exception:
Traceback (most recent call last):
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\cli\base_command.py", line 143, in main
    status = self.run(options, args)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\commands\install.py", line 318, in run
    resolver.resolve(requirement_set)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\resolve.py", line 256, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\resolve.py", line 209, in _get_abstract_dist_for
    self.require_hashes
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\operations\prepare.py", line 283, in prepare_linked_requirement
    progress_bar=self.progress_bar
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 836, in unpack_url
    progress_bar=progress_bar
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 673, in unpack_http_url
    progress_bar)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 897, in _download_http_url
    _download_url(resp, link, content_file, hashes, progress_bar)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 619, in _download_url
    consume(downloaded_chunks)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\utils\misc.py", line 848, in consume
    deque(iterator, maxlen=0)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 585, in written_chunks
    for chunk in chunks:
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\utils\ui.py", line 159, in iter
    for x in it:
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_internal\download.py", line 574, in resp_read
    decode_content=False):
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 465, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 430, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "C:\Users\Antec\.platformio\python27\Lib\contextlib.py", line 35, in __exit__
    self.gen.throw(type, value, traceback)
  File "C:\Users\Antec\.platformio\penv\lib\site-packages\pip\_vendor\urllib3\response.py", line 345, in _error_catcher
    raise ReadTimeoutError(self._pool, None, 'Read timed out.')
ReadTimeoutError: HTTPSConnectionPool(host='codeload.github.com', port=443): Read timed out.

 If you have antivirus software in a system, try to disable it for a while.
	at i.runCommand (C:\Users\Antec\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:26030)
	at ChildProcess.h (C:\Users\Antec\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:5944)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\Antec\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ivankravets, 2018-11-21:
1. Do you use proxy?
2. Do you have antivirus?

Antecer, 2018-11-21:
I already fixed, just stop  Windows Defender Antivirus.

ivankravets, 2018-11-21:
Thanks! I added better explanation https://github.com/platformio/platformio-node-helpers/commit/e2093d09465bc6a1c74fe9363b7dd112e3fe849b

closed 2018-11-21 by ivankravets