platformio/platformio-vscode-ide #425: Start PIO Home Server

AxeMeterHou, 2018-11-21:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.5
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 37, in <module>
Exception: PIO Plus: Runtime error

	at C:\Users\后涛\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:15518
	at ChildProcess.h (C:\Users\后涛\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:5944)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\后涛\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

smuehlst, 2019-04-21:
Many people apparently run into this, and so do l. It looks like having a non-ASCII character in the HOME directory path triggers this problem.

Is there any workaround? I tried to specify a different location for the .vscode folder of Visual Studio Code with the command line option "--extensions-folder", but without success.

ivankravets, 2019-05-28:
Fixed in the latest version of PIO Core 4.0. Please switch to a development version of PIo Core http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2019-07-13:
Resolve in the latest PIO Core 4.0. Please upgrade to PIO Core 4.0

closed 2019-07-13 by ivankravets