platformio/platformio-vscode-ide #426: Installation Manager

kopfschmerz1, 2018-11-22:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.5
System: Windows_NT, 10.0.15063, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv
	at t.value (C:\Users\rutsaral\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:25260)
	at <anonymous>
```

ivankravets, 2018-11-23:
Please post here the errors from VSCode > Menu > Help > Toggle Developer Tools > Console

kopfschmerz1, 2018-11-23:
It is already all right! Thank yoy very much for your attention!
BR,
Alexander
--
Отправлено из Mail.

ivankravets, 2018-11-23:
How did you resolve it?

kopfschmerz1, 2018-11-23:
Probably i was behind the firewall. So after reconnect to another network, the problem was resolved.
--
Отправлено из Mail.

ivankravets, 2018-11-23:
Thanks! Happy coding with PlatformIO!

closed 2018-11-23 by ivankravets