platformio/platformio-vscode-ide #429: Installation Manager

FaKevin, 2018-11-23:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.5
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv
	at t.value (C:\Users\Fabricio Merida\.vscode\extensions\platformio.platformio-ide-1.4.5\node_modules\platformio-node-helpers\dist\index.js:1:25260)
	at <anonymous>
```

ivankravets, 2018-11-23:
Please post here the errors from VSCode > Menu > Help > Toggle Developer Tools > Console

FaKevin, 2018-11-25:
`Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.     Cache entry deserialization failed, entry ignored C:\Program Files\QGIS 2.18\apps\Python27\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py:339: SNIMissingWarning: An HTTPS request has been made, but the SNI (Subject Name Indication) extension to TLS is not available on this platform. This may cause the server to present an incorrect TLS certificate, which can cause validation failures. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings   SNIMissingWarning C:\Program Files\QGIS 2.18\apps\Python27\lib\site-packages\pip\_vendor\urllib3\util\ssl_.py:137: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Pyth...`

closed 2018-11-25 by FaKevin

ivankravets, 2018-11-25:
What was a Python version?