platformio/platformio-vscode-ide #434: Your VS Code extension is affected by a security issue.

auchenberg, 2018-11-28:
Hi,

Kenneth here from the VS Code team. 

Your extension is affected by https://code.visualstudio.com/blogs/2018/11/26/event-stream, as we have detected you are using `flatmap-stream v0.1.2`.

We will need to take action to block your extension if your extension isn't updated in a reasonable timeframe.

Please get in touch at kennau@microsoft.com

/k

ivankravets, 2018-11-28:
Hi Kenneth,

Thank you so much for reporting this issue. I've just done some research on this issue and don't see any problems with our extension. It is not affected. Our production dependencies don't depend on `event-stream`. The only one package which depends is `vscode`. it depends on different `gulp-*` packages which use vulnerable `event-stream`.

[image]

[image]

Yes, I see this vulnerable package in our VSIX extension file but it just a development dependency of `vscode`. The entry point declared in `package.json` does not use `event-stream`.

--------

I also don't understand why `$ vsce package` pack DEVELOPMENT packages. Why do we need them? our output VSIX archive has 5Mb size and we don't need 4.99Mb of node's dev packages. Could you forward this issue to your internal team? If you resolve this issue, then we can avoid similar security problems in future.

Thanks!

ivankravets, 2018-11-28:
The new release is out and published in market place! https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.4.7

auchenberg, 2018-11-28:
Wonderful. Thanks!

closed 2018-11-28 by auchenberg