platformio/platformio-vscode-ide #436: Installation Manager

obaidaattaee, 2018-11-28:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.6
System: Linux, 4.18.0-parrot10-686-pae, ia32

# Exception
```
Error: Failed to download file https://files.pythonhosted.org/packages/45/ae/8a0ad77defb7cc903f09e551d88b443304a9bd6e6f124e75c0fbbf6de8f7/pip-18.1.tar.gz: Error: getaddrinfo EAI_AGAIN files.pythonhosted.org:443
	at Object.t.download (/home/obaida/.vscode/extensions/platformio.platformio-ide-1.4.6/node_modules/platformio-node-helpers/dist/index.js:1:29236)
	at <anonymous>
	at process._tickCallback (internal/process/next_tick.js:188:7)
```

ivankravets, 2018-11-29:
Do you use proxy?

ivankravets, 2018-11-29:
Fixed in https://github.com/platformio/platformio-node-helpers/commit/44473ecf730c689641f612693f07c7e28a013914

commit 2e4d3ef3 (2018-11-29): Update PlatformIO Core Installer // Resolve #436, Issue #154
closed 2018-11-29 by ivankravets