platformio/platformio-vscode-ide #437: Installation Manager

KeithWatterson, 2018-11-28:
# Description of problem
Having problems. I could not log in to PIO Plus, it kept telling me there was no internet connection.
So I gave up and installed the package from VSCode and deleted the ~/.platformio/penv folder.
Then from VSCode reinstall the Platform io package and this is the result

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.4.7
System: Darwin, 18.2.0, x64

# Exception
```
Error: PIP Core:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b91390>: Failed to establish a new connection: [Errno 64] Host is down',)': /platformio/platformio-core/archive/develop.zip
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b914d0>: Failed to establish a new connection: [Errno 64] Host is down',)': /platformio/platformio-core/archive/develop.zip
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b91610>: Failed to establish a new connection: [Errno 64] Host is down',)': /platformio/platformio-core/archive/develop.zip
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b91750>: Failed to establish a new connection: [Errno 64] Host is down',)': /platformio/platformio-core/archive/develop.zip
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b91890>: Failed to establish a new connection: [Errno 64] Host is down',)': /platformio/platformio-core/archive/develop.zip
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio-core/archive/develop.zip (Caused by NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x104b91990>: Failed to establish a new connection: [Errno 64] Host is down',))

	at i.runCommand (/Users/Keith/.vscode/extensions/platformio.platformio-ide-1.4.7/node_modules/platformio-node-helpers/dist/index.js:1:26260)
	at ChildProcess.h (/Users/Keith/.vscode/extensions/platformio.platformio-ide-1.4.7/node_modules/platformio-node-helpers/dist/index.js:1:6004)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at maybeClose (internal/child_process.js:925:16)
	at Socket.stream.socket.on (internal/child_process.js:346:11)
	at emitOne (events.js:116:13)
	at Socket.emit (events.js:211:7)
	at Pipe._handle.close [as _onclose] (net.js:554:12)
```

KeithWatterson, 2018-11-28:
oops meant to sy "So I gave up and un-installed the package from VSCode

ivankravets, 2018-11-29:
Please update to the latest 1.5.0 release and restart VSCode. Does it work now?

- https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.5.0

KeithWatterson, 2018-11-30:
Ivan
I think it was ‘little snitch’ blocking all viscode internet access.
regard
Keith

KeithWatterson, 2018-11-30:
Yes Thanks

I don't know why Little Sniched did what it did, so I told it not to block VCODE and all is well
Keith

ivankravets, 2018-11-30:
Happy coding with PlatformIO!

closed 2018-11-30 by ivankravets