platformio/platformio-vscode-ide #438: Include file not found in browse.path

KarateBrot, 2018-11-28:
A few hours ago this warning popped up without changing anything in my code structure. So I checked the `c_cpp_properties.json` of my project and all the error is right there: 

**all links in browse.path have an additional `}` at the end** so all links are invalid as you can see:

```javascript
"browse": {
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": "${workspaceRoot}/.vscode/.browse.c_cpp.db",
                "path": [
                    "C:/Users/Jan Post/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include}",
                    "C:/Users/Jan Post/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include}",
                    "C:/Users/Jan Post/.platformio/packages/framework-arduinoespressif8266/cores/esp8266}",
                    ...
```
This just looks like a minor typing error in the function that generates the path strings

ivankravets, 2018-11-28:
Could you try to remove `.vscode` folder and restart VSCode?

KarateBrot, 2018-11-28:
I just deleted it but after restarting vscode a new faulty file with gets created.

ivankravets, 2018-11-29:
This is a bug and was fixed in https://github.com/platformio/platformio-core/commit/16df5474e4a3d5d2edc9324c6dca8b7db84a450d

Please run Left SiderBar > PlatformIO > Quick Access > Upgrade PIO Core

closed 2018-11-29 by ivankravets

MHz000, 2020-04-07:
I've still the same problem after ugrade PIO core:
-----------------------------------------------------
PS C:\Users\Win10\Documents\PlatformIO\Projects\ESP_ballet> platformio upgrade
You're up-to-date!
PlatformIO 4.3.1 is currently the newest version available.
PS C:\Users\Win10\Documents\PlatformIO\Projects\ESP_ballet> 
-----------------------------------------------------
I'm just at the beginning to use/discover platformio. Pls be patient and give detailed instructions.
Thank you