platformio/platformio-vscode-ide #441: Installation Manager

mmphego, 2018-11-30:
# Description of problem
Please see the exception below. if it isn't a legitimate bug close the issue.
Thanks

# Configuration

VSCode: 1.29.1
PIO IDE: v1.5.0
System: Linux, 4.15.0-39-generic, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv.py", line 2462, in <module>
    main()
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv.py", line 1015, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv.py", line 968, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv.py", line 854, in call_subprocess
    raise OSError("Command {} failed with error code {}".format(cmd_desc, proc.returncode))
OSError: Command /home/mmphego/.platformio/penv/bin/python2.7 - setuptools pip wheel failed with error code 1

New python executable in /home/mmphego/.platformio/penv/bin/python2.7
Also creating executable in /home/mmphego/.platformio/penv/bin/python
Installing setuptools, pip, wheel...

  Complete output from command /home/mmphego/.platformio/penv/bin/python2.7 - setuptools pip wheel:
  Exception:
Traceback (most recent call last):
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/commands/install.py", line 259, in run
    with self._build_session(options) as session:
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/cli/base_command.py", line 76, in _build_session
    if options.cache_dir else None
  File "/home/mmphego/.platformio/penv/lib/python2.7/posixpath.py", line 77, in join
    elif path == '' or path.endswith('/'):
AttributeError: 'int' object has no attribute 'endswith'
Traceback (most recent call last):
  File "<stdin>", line 27, in <module>
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/__init__.py", line 78, in main
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/cli/base_command.py", line 184, in main
  File "/home/mmphego/.platformio/.cache/tmp-21408XNLxzw0Zqtc5/virtualenv-16.1.0/src/virtualenv_support/pip-18.1-py2.py3-none-any.whl/pip/_internal/cli/base_command.py", line 76, in _build_session
  File "/home/mmphego/.platformio/penv/lib/python2.7/posixpath.py", line 77, in join
    elif path == '' or path.endswith('/'):
AttributeError: 'int' object has no attribute 'endswith'
----------------------------------------
...Installing setuptools, pip, wheel...done.

    at t.value (/home/mmphego/.vscode/extensions/platformio.platformio-ide-1.5.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/installer/stages/platformio-core.js:278:21)
    at <anonymous>
```

ivankravets, 2018-11-30:
Could you try to install `python2-virtualenv` package and restart VSCode?

mmphego, 2018-12-03:
Seems like it is my current system.
I'll close this ticket and reinstall my setup.

closed 2018-12-03 by mmphego