platformio/platformio-vscode-ide #443: Installation Manager

RiPeter, 2018-12-01:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.5.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Failed to download file https://files.pythonhosted.org/packages/4e/8b/75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690/virtualenv-16.1.0.tar.gz: Error: EPERM: operation not permitted, open 'C:\Users\peter\.platformio\.cache\virtualenv.tar.gz'
	at t.value (C:\Users\peter\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\platformio-node-helpers\dist\index.js:1:25327)
	at <anonymous>
```

ivankravets, 2018-12-12:
> Error: EPERM: operation not permitted

Do you use a proxy? Do you have a firewall? Please check your internet connection.

RiPeter, 2018-12-12:
Thank you! Problem solved.

ivankravets, 2018-12-12:
How did you resolve this issue?

closed 2018-12-12 by ivankravets

RiPeter, 2018-12-12:
I restarted the program. Thank you!

> 2018. dec. 12. dátummal, 11:33 időpontban Ivan Kravets <notifications@github.com> írta:
> 
> How did you resolve this issue