platformio/platformio-vscode-ide #450: Start PIO Home Server

yurybotov, 2018-12-05:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.5.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Error: Could not load broken JSON: C:\Users\Yury\.platformio\packages\tool-pioplus\package.json

	at C:\Users\Yury\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\platformio-node-helpers\dist\index.js:1:14992
	at ChildProcess.h (C:\Users\Yury\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\Yury\.vscode\extensions\platformio.platformio-ide-1.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ivankravets, 2018-12-12:
Please remove `C:\Users\Yury\.platformio\packages\tool-pioplus` folder and disable antivirus for a while. Restart VSCode. It seems that antivirus blocks PlatformIO's package manager.

closed 2018-12-12 by ivankravets

ivankravets, 2018-12-12:
Happy coding with PlatformIO!