platformio/platformio-vscode-ide #459: VS Code, Intellisense does not work but code compiles, uploads and stores

ccourson, 2018-12-09:
I've been working with mbed and STM32 devices. On VS Code, Intellisense does not work but code compiles, uploads and stores. Worked after first install but over time with updates broke. I have tried solutions mentioned in this area without success. As suggested in reply to my review, I have removed all other extensions from VS Code.

[image]

ivankravets, 2018-12-11:
Fixed in https://github.com/platformio/platformio-core/commit/f5395133767c9d5f0c7f0d703965006c981a8334

Please enable development verison of built-in PIO Core. Set `platformio-ide.useDevelopmentPIOCore` to `true` in https://docs.platformio.org/en/latest/ide/vscode.html#settings

Restart VSCode. Does it work now?

ivankravets, 2018-12-11:
Sorry for the delay, we have been at RISC-V Summit https://twitter.com/PlatformIO_Org/status/1071111444718993408

ivankravets, 2018-12-12:
Could you share your `.vscode/c_cpp_properties.json` file from project folder? It seems that you have another issue.

closed 2018-12-12 by ccourson

ivankravets, 2018-12-12:
> "pio upgrade --dev" was the additional magic

it switched PlatformIO Core to the latest 3.6.3b2. We will have the next release soon.

ivankravets, 2018-12-12:
Would be thankful if you update your review https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide#review-details

Sorry for the delay again.

ccourson, 2018-12-12:
Done. Thank you very much. Looking forward to working with other devices using PlatformIO.

ivankravets, 2018-12-12:
Thank you too for the great feedback! 😊 

Happy coding with PlatformIO!

commit 09d569bb (2018-12-12): Fix spurious project's "Problems" when ARM mbed framework is used // Resolve #459