platformio/platformio-vscode-ide #473: Update PIO Home Content

jcsjesus, 2018-12-16:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.0
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not start PIO Home server: Error: connect EADDRNOTAVAIL 127.0.0.1 - Local (0.0.0.0:59260)
	at r.P.l.default.waitUntilUsed.then.e (C:\Users\José Carlos\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:15068)
	at <anonymous>
	at process._tickCallback (internal/process/next_tick.js:188:7)
```

jcsjesus, 2018-12-16:
Do not download any models, no libraries, or BOARDS NOTHING

ivankravets, 2019-01-30:
See https://github.com/platformio/platformio-vscode-ide/issues/457#issuecomment-446623752

closed 2019-01-30 by ivankravets