platformio/platformio-vscode-ide #474: Installation Manager

HannHank, 2018-12-16:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Johann Hanke\.platformio\.cache\tmp-4112H4PjmPP97qsC\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\Johann Hanke\.platformio\.cache\tmp-4112H4PjmPP97qsC\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\Johann Hanke\.platformio\.cache\tmp-4112H4PjmPP97qsC\virtualenv-16.1.0\src\virtualenv.py", line 998, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\Johann Hanke\.platformio\.cache\tmp-4112H4PjmPP97qsC\virtualenv-16.1.0\src\virtualenv.py", line 1317, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\Johann Hanke\.platformio\python27\Lib\shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: 'C:\\Users\\JOHANN~1\\PLATFO~1\\penv\\Scripts\\python.exe'

New python executable in C:\Users\JOHANN~1\PLATFO~1\penv\Scripts\python.exe

	at t.value (C:\Users\Johann Hanke\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:25327)
	at <anonymous>
```

tovine, 2019-01-02:
Got exactly the same issue now, in my case it happens when I start VSCode and it has more than one window open from the last session.
I'm pretty sure it has something to do with concurrency because all windows attempt to run the same installation/check and the first one locks the file so the others fail.

For me I had 3 instances open at the same time; one succeeded and the two others failed. I suggest using a temporary lockfile or similar, to indicate that an installation is in progress; then the existence of that file should be checked before attempting to start a new installation process.

ivankravets, 2019-01-30:
We have a lock, see https://github.com/platformio/platformio-vscode-ide/blob/develop/src/installer/manager.js#L50

Could you reproduce this issue again? Why does PlatformIO reinstall itself each time?

tovine, 2019-01-31:
Unable to reproduce now, it may have been because of updates to VSCode (it has usually been the reason why I've restarted the application before), or perhaps updates to PlatformIO.

ivankravets, 2019-01-31:
Please reopen if you will experience this issue.

closed 2019-01-31 by ivankravets