platformio/platformio-vscode-ide #478: Extension causes high cpu load

pegasussan, 2018-12-19:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/2694850/platformio.platformio-ide-unresponsive.cpuprofile.txt)
- Issue Type: `Performance`
- Extension Name: `platformio-ide`
- Extension Version: `1.6.0`
- OS Version: `Linux x64 4.15.0-34-generic`
- VSCode version: `1.30.0`

:warning: Make sure to **attach** this file from your *home*-directory: `/home/pegasus/platformio.platformio-ide-unresponsive.cpuprofile.txt` :warning:

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

cabletie, 2019-06-10:
Not sure if I'm doing the right thing here but it seems this issue is still open and you're after these dump files for more info.

Apologies if I'm cluttering up your issue log ...

[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/3271793/platformio.platformio-ide-unresponsive.cpuprofile.txt)

- cabletie

leoesilva, 2019-09-16:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/3618006/platformio.platformio-ide-unresponsive.cpuprofile.txt)

DivingDuck, 2019-10-04:
I have the same issue after I integrate plattformio to VSCode. Please find attached log below. Any help is appreciated.
Best regards, DD

[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/3691848/platformio.platformio-ide-unresponsive.cpuprofile.txt)

wisegeeksrules, 2020-04-04:
I had the same issue, check the text below:

[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/4431150/platformio.platformio-ide-unresponsive.cpuprofile.txt)

iftoz, 2020-06-21:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/4808778/platformio.platformio-ide-unresponsive.cpuprofile.txt)
hello there, I have the same problem here running on my windows 7 64bit, could you please suggest which case that soled the same issue as mine?

Thank you guys, you all are really working on something great :)

eduardodarocha, 2020-08-12:
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/5064879/platformio.platformio-ide-unresponsive.cpuprofile.txt)
Hello I have this problem . Can you help me to find a solution? 
Thanks

closed 2020-09-21 by stale[bot]