platformio/platformio-vscode-ide #479: Opened VSCode since 3 weeks and PlatformIO update crashes

moracabanas, 2018-12-19:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.

Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\morac\.platformio\.cache\tmp-8332kF9ef011VUmj\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\morac\.platformio\.cache\tmp-8332kF9ef011VUmj\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\morac\.platformio\.cache\tmp-8332kF9ef011VUmj\virtualenv-16.1.0\src\virtualenv.py", line 1015, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\morac\.platformio\.cache\tmp-8332kF9ef011VUmj\virtualenv-16.1.0\src\virtualenv.py", line 968, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\morac\.platformio\.cache\tmp-8332kF9ef011VUmj\virtualenv-16.1.0\src\virtualenv.py", line 812, in call_subprocess
    env=env,
  File "C:\Users\morac\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\morac\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Acceso denegado

Please use the *system* python to run this script
Installing setuptools, pip, wheel...

  Error [Error 5] Acceso denegado while executing command "C:\Program Files\Microsoft VS Code" - setuptools pip wheel
...Installing setuptools, pip, wheel...done.

	at t.value (C:\Users\morac\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:25327)
	at <anonymous>
```

JanJohansen, 2019-01-18:
Same here...

Version: 1.30.2 (user setup)
Commit: 61122f88f0bf01e2ac16bdb9e1bc4571755f5bd8
Date: 2019-01-07T22:54:13.295Z
Electron: 2.0.12
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
OS: Windows_NT x64 10.0.17134

ivankravets, 2019-01-30:
Please create it manually -> http://bit.ly/pio-core-virtualenv

Do you have antivirus tools in a system? Try to disable them while installing IDE or install virtual environment manually => http://bit.ly/pio-core-virtualenv

closed 2019-01-30 by ivankravets