platformio/platformio-vscode-ide #522: Start PIO Home Server

yifanff, 2018-12-26:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.0
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: 
	at r.P.Promise (C:\Users\liyifan\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\liyifan\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\liyifan\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.__dirname.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

ivankravets, 2019-01-30:
Could you provide more details? VSCode > Help > Toggle Developer Tools > Console

Do you see any errors?

ivankravets, 2019-02-22:
See https://github.com/platformio/platformio-vscode-ide/issues/573

closed 2019-02-22 by ivankravets