platformio/platformio-vscode-ide #523: Start PIO Home Server

HannHank, 2018-12-26:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 35, in <module>
  File "<string>", line 15, in load_module
  File "<string>", line 2, in <module>
  File "C:\Users\Johann Hanke\.platformio\packages\contrib-pysite\twisted\internet\threads.py", line 19, in <module>
    from twisted.internet import defer
  File "C:\Users\Johann Hanke\.platformio\packages\contrib-pysite\twisted\internet\defer.py", line 21, in <module>
    import attr
  File "C:\Users\Johann Hanke\.platformio\packages\contrib-pysite\attr\__init__.py", line 5, in <module>
    from . import converters, exceptions, filters, validators
  File "C:\Users\Johann Hanke\.platformio\packages\contrib-pysite\attr\filters.py", line 8, in <module>
    from ._make import Attribute
  File "C:\Users\Johann Hanke\.platformio\packages\contrib-pysite\attr\_make.py", line 21, in <module>
    from .exceptions import (
ImportError: No module named exceptions

	at C:\Users\Johann Hanke\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992
	at ChildProcess.h (C:\Users\Johann Hanke\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\Johann Hanke\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

kasra0, 2018-12-27:
Hello,

i have the same issue.

My configuration is : 
VSCode: 1.30.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

Thanks

kasra0, 2018-12-27:
hi,
i closed vscode.
i deleted the folder C:\Users\xxx\.platformio.
I opened vscode, the ide launched automatically the installation of PIO.
I realoded then the ide and  was able to load the plugin.

ivankravets, 2019-01-30:
Please remove `C:\Users\Johann Hanke\.platformio\packages\contrib-pysite` folder and restart VSCode.

closed 2019-01-30 by ivankravets