platformio/platformio-vscode-ide #525: serial monitor traffic not appearing until board reset: how to debug?

mbariola, 2018-12-27:
I have a weird machine-dependent behavior that I cannot troubleshoot so far.

I am running vscode 1.30.1 and platformIO 2.0.0 w/core 3.6.4b1 on Windows 10, on 2 different machines.
Settings are kept synced via a vscode extension.

On the laptop, I upload & monitor some code and I can see its output in the serial monitor.
When I upload and monitor the same code (synced through dropbox so exact same codebase) on my desktop computer, serial monitor opens but no data is shown until I hard reset the board with its RST button. Board is the same for both tests.

Any ideas of why, or how to troubleshoot this?
thanks

ivankravets, 2019-01-30:
What is your board?

mbariola, 2019-02-27:
Sorry, had lost your reply.
I am using [this Lolin board](https://www.aliexpress.com/item/Lolin-ESP32-OLED-V2-0-Pro-ESP32-OLED-wemos-pour-Arduino-ESP32-OLED-WiFi-Modules-Bluetooth/32822105291.html?spm=a2g0s.9042311.0.0.3da24c4duj7i3z)

with the following `platformio.ini` bit:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  ESP32 BLE Arduino
  ESP8266_SSD1306
monitor_speed = 115200
```

ivankravets, 2019-03-01:
Do you use the same driver TTL/UART converter?

mbariola, 2019-03-01:
if talking software, I am not sure how to check this - had a look at device manager but didn't find anything helpful. How can I check it?

ivankravets, 2019-07-23:
Please move this issue to https://community.platformio.org/, it seems like a very specific case. Maybe, the community will help.

closed 2019-07-23 by ivankravets

mbariola, 2019-07-23:
Hi, unfortunately I cannot replicate right now as I am not presently in possession of the machine where the behavior happened. I'll open a new ticket on the community if problem persists. thanks