platformio/platformio-vscode-ide #542: Could not start PIO home in VS studio-ide and Atom

paul5196, 2019-01-08:


paul5196, 2019-01-08:
Platform is Windows 1803 x64 bit.
using VScode  and opening the platformio ide I receive an error after about 2 minutes  the error reads 👍

 Error: Traceback (most recent call last): File "<string>", line 1, in <module> File "<string>", line 38, in <module> File "<string>", line 38, in <module> File "<string>", line 18, in load_module File "<string>", line 9, in <module> File "C:\Users\yyy\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <module> from twisted.internet._win32serialport import SerialPort File "C:\Users\yyy\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, in <module> import win32file, win32event ImportError: DLL load failed: The network name cannot be found. at r.P.Promise (C:\Users\yyy\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992) at ChildProcess.h (C:\Users\yyy\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916) at emitTwo…

If I open a terminal I cannot run PIO:

PS C:\Users\yyy> pio home
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable
program. Check the spelling of the name, or if a path was included, verify that the path is correct and
try again.
At line:1 char:1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

If I now try the same with ATOM ide I get

Error: Traceback (most recent call last): File "", line 1, in File "", line 38, in File "", line 38, in File "", line 18, in load_module File "", line 9, in File "C:\Users\yyy.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in from twisted.internet._win32serialport import SerialPort File "C:\Users\yyy.platformio\packages\contrib-pysite\twisted\internet_win32serialport.py", line 19, in import win32file, win32event ImportError: DLL load failed: The network name cannot be found.

at C:\Users\yyy\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:14992
at ChildProcess.h (C:\Users\yyy\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\dist\index.js:1:5916)
at emitTwo (events.js:126:13)
at ChildProcess.emit (events.js:214:7)
at ChildProcess.cp.emit (C:\Users\yyy\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
at maybeClose (internal/child_process.js:925:16)
at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)

And if I run a terminal I get:

PS C:\Users\ahepa01.revotechnik> pio home
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 9, in <module>
  File "C:\Users\yyy\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <mod
ule>
    from twisted.internet._win32serialport import SerialPort
  File "C:\Users\yyy\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, i
n <module>
    import win32file, win32event
ImportError: DLL load failed: The network name cannot be found.

I am at a loss. I have this functional on a PC at home same Windows OS and updates and on a Linux box in the office. but not on this PC. I may add this PC has Visual Studio Professional on it - could that prevent the operation of PlatformIO?

ivankravets, 2019-01-11:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

paul5196, 2019-01-11:
Outputs

C:\Users\ahepa01.REVOTECHNIK>echo %PATH%
\\PLUTUS\sage300\RUNTIME;\\10.1.0.201\sage300\RUNTIME;\\10.1.0.200\sage300\RUNTIME;C:\Program Files (x86)\Intel\iCLS Cli
ent\;C:\Program Files\Intel\iCLS Client\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\
Java\javapath;C:\Program Files (x86)\Microchip\mplabc18\v3.47\mpasm;C:\Program Files (x86)\Microchip\mplabc18\v3.47\bin;
C:\MCC18\mpasm;C:\MCC18\bin;\\154.20.5.20\sage300\RUNTIME;C:\Program Files\Common Files\Microsoft Shared\Microsoft Onlin
e Services;C:\Program Files (x86)\Common Files\Microsoft Shared\Microsoft Online Services;C:\Program Files\Microsoft MPI
\Bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;D:\Support Too
ls\;C:\Program Files (x86)\GNU\GnuPG\pub;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\dotnet\;
C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files (x86)\Microchip\xc32\v1.43\bin;C:\Program Files (x86)\Microchi
p\MPLAB C32 Suite\bin;C:\Program Files\SafeNet\Authentication\SAC\x64;C:\Program Files\SafeNet\Authentication\SAC\x32;C:
\Program Files\PuTTY\;C:\Program Files (x86)\Microchip\xc16\v1.33\bin;C:\Program Files (x86)\Microchip\xc8\v1.42\bin;C:\
Program Files (x86)\Microchip\xc8\v1.45\bin;C:\Program Files (x86)\Nordic Semiconductor\nrf5x\bin\;C:\Program Files (x86
)\Microchip\xc16\v1.34\bin;C:\Program Files\doxygen\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Componen
ts\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management
Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\System32\OpenSSH\;C:\
Program Files\Git\cmd;C:\Program Files (x86)\WinMerge;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\Micr
ochip\mplabc18\v3.47\mpasm;C:\Program Files (x86)\Microchip\mplabc18\v3.47\bin;C:\MCC18\mpasm;C:\MCC18\bin;C:\Users\ahep
a01.REVOTECHNIK\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Microchip\xc8\v1.42\bin;C:\Program Files (x86
)\Nmap;"C:\Python27;";"C:\Python27\Scripts;";"C:\Users\ahepa01.REVOTECHNIK\.platformio\penv\Scripts;";C:\Users\ahepa01.R
EVOTECHNIK\AppData\Local\Microsoft\WindowsApps;

C:\Users\ahepa01.REVOTECHNIK>echo
%PYTHONPATH%

C:\Users\ahepa01.REVOTECHNIK>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

It appears I do not have a path to PYTHON

But python27 is on the system
[cid:image002.png@01D4A9B5.5E58A730]

Do I just create a path variable  and add the path?

Kind regards
Paul Ahern.     B.Eng(Hons), I.Eng, MIET, MIEEE
Senior Applications Engineer
Revo Developments Ltd

paul5196, 2019-01-11:
So I have edited the path by adding a PATH variable PYTHONPATH = C:\Pyhton27;C:\Python27\Lib\site-packages;C:\Python27\Scripts

Output from the echo %PATH%

C:\Users\ahepa01.REVOTECHNIK>echo %PATH%
\\PLUTUS\sage300\RUNTIME;\\10.1.0.201\sage300\RUNTIME;\\10.1.0.200\sage300\RUNTIME;C:\Program Files (x86)\Intel\iCLS Cli
ent\;C:\Program Files\Intel\iCLS Client\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\
Java\javapath;C:\Program Files (x86)\Microchip\mplabc18\v3.47\mpasm;C:\Program Files (x86)\Microchip\mplabc18\v3.47\bin;
C:\MCC18\mpasm;C:\MCC18\bin;\\154.20.5.20\sage300\RUNTIME;C:\Program Files\Common Files\Microsoft Shared\Microsoft Onlin
e Services;C:\Program Files (x86)\Common Files\Microsoft Shared\Microsoft Online Services;C:\Program Files\Microsoft MPI
\Bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;D:\Support Too
ls\;C:\Program Files (x86)\GNU\GnuPG\pub;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\dotnet\;
C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files (x86)\Microchip\xc32\v1.43\bin;C:\Program Files (x86)\Microchi
p\MPLAB C32 Suite\bin;C:\Program Files\SafeNet\Authentication\SAC\x64;C:\Program Files\SafeNet\Authentication\SAC\x32;C:
\Program Files\PuTTY\;C:\Program Files (x86)\Microchip\xc16\v1.33\bin;C:\Program Files (x86)\Microchip\xc8\v1.42\bin;C:\
Program Files (x86)\Microchip\xc8\v1.45\bin;C:\Program Files (x86)\Nordic Semiconductor\nrf5x\bin\;C:\Program Files (x86
)\Microchip\xc16\v1.34\bin;C:\Program Files\doxygen\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Componen
ts\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management
Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\System32\OpenSSH\;C:\
Program Files\Git\cmd;C:\Program Files (x86)\WinMerge;C:\Program Files\Microsoft VS Code\bin;\\PLUTUS\sage300\RUNTIME;\\
10.1.0.201\sage300\RUNTIME;\\10.1.0.200\sage300\RUNTIME;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel
\iCLS Client\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Fi
les (x86)\Microchip\mplabc18\v3.47\mpasm;C:\Program Files (x86)\Microchip\mplabc18\v3.47\bin;C:\MCC18\mpasm;C:\MCC18\bin
;\\154.20.5.20\sage300\RUNTIME;C:\Program Files\Common Files\Microsoft Shared\Microsoft Online Services;C:\Program Files
(x86)\Common Files\Microsoft Shared\Microsoft Online Services;C:\Program Files\Microsoft MPI\Bin\;C:\WINDOWS\system32;C
:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;D:\Support Tools\;C:\Program Files (x86)\
GNU\GnuPG\pub;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\dotnet\;C:\Program Files (x86)\GtkS
harp\2.12\bin;C:\Program Files (x86)\Microchip\xc32\v1.43\bin;C:\Program Files (x86)\Microchip\MPLAB C32 Suite\bin;C:\Pr
ogram Files\SafeNet\Authentication\SA;C:\Pyhton27;C:\Python27\Lib\site-packages;C:\Python27\Scripts

Output from the  echo %PYTHONPATH%

C:\Users\ahepa01.REVOTECHNIK>echo %PYTHONPATH%
C:\Python27;C:\Python27\Lib\site-packages;C:\Python27\Scripts

Output from python –version

C:\Users\ahepa01.REVOTECHNIK>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

But….
C:\Users\ahepa01.REVOTECHNIK>C:\python27\python --version
Python 2.7.13

And

C:\Users\ahepa01.REVOTECHNIK>C:\python27\Scripts\pip --version
pip 18.1 from C:\Python27\Lib\site-packages\pip (python 2.7)

So this clearly indicates that the path is not set. I have reset the machine and I have also rebooted the machine but it still refuses to set the path? Even though the path is clearly visible from the above tests.

Kind regards
Paul Ahern.     B.Eng(Hons), I.Eng, MIET, MIEEE
Senior Applications Engineer
Revo Developments Ltd

ivankravets, 2019-01-11:
1. Please uninstall all Pythons from a system and cleanup system PATH from any Python paths
2. Remove `C:\Users\yyy\.platformio` folder
3. Start VSCode

Does it work now?

ivankravets, 2019-02-22:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/586

closed 2019-02-22 by ivankravets