platformio/platformio-vscode-ide #543: Installation Manager

fbigun, 2019-01-08:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.29.0
PIO IDE: v1.6.0
System: Linux, 4.15.0-29deepin-generic, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Failed to download file https://files.pythonhosted.org/packages/4e/8b/75469c270ac544265f0020aa7c4ea925c5284b23e445cf3aa8b99f662690/virtualenv-16.1.0.tar.gz: Error: tunneling socket could not be established, cause=Parse Error
	at t.value (/home/olutyo/.vscode/extensions/platformio.platformio-ide-1.6.0/node_modules/platformio-node-helpers/dist/index.js:1:25327)
	at <anonymous>
```

ivankravets, 2019-01-10:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

doubledutch2, 2020-09-07:
I have exactly the same issue on a brand new Mac. PYTHONPATH is empty, Python version is 2.7 whilst I have python3 installed as well and pip is not installed either. Any suggestions?

ivankravets, 2020-09-08:
@doubledutch2 could you try the upcoming PlatformIO IDE 2.0? 
- https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta-help-us-with-testing/14097

Does it work for you?

doubledutch2, 2020-09-08:
> @doubledutch2 could you try the upcoming PlatformIO IDE 2.0?

@ivankravets - I did try that as well and get exactly the same issue. It must be something on my mac but I have no idea what I'm missing. The mac is brand new, I just installed VSC and PlatformIO is the first and only extension. The only thing that I can have happened is that I use TimeMachine to move my old content to my new mac so maybe that messed something up. I (tried to) delete all VSC and PlatformIO stuff and install again but that made no difference.

ivankravets, 2020-09-08:
@doubledutch2 could you mail me@ikravets.com? I'd like to debug this issue. We are goging to release IDE 2.0 this week and it would be great to fix your issue.