platformio/platformio-vscode-ide #545: Override upload options in the IDE

jeremypoulter, 2019-01-09:
In a similar vain to #544 the other change I find my self making in the `platformio.ini` files, that shouldn't be committed, is the upload settings. Mainly when doing ESP8266 dev I sometimes need to make a change to use a different host name.

Would be nice if the available `upload_port` options could be selected from a list as in the Arduino IDE, but I would be happy with manually typing them.

`upload_speed` would also be handy to be to set but rarely have to change that.

closed 2022-12-18 by ivankravets

ivankravets, 2022-12-18:
Implemented in the latest development version (upcoming PlatformIO IDE 3.0). Please re-test with the pre-release version. See the instructions below:

[image]

dzid26, 2023-02-17:
Does a similar exist for `upload_protocol`?

commit 864706e5 (2022-12-18): Override upload/monitor/test port with a new port switcher // Resolve #545