platformio/platformio-vscode-ide #551: Extension causes high cpu load

scopelemanuele, 2019-01-12:
- Issue Type: `Performance`
- Extension Name: `platformio-ide`
- Extension Version: `1.6.0`
- OS Version: `Linux x64 4.20.0-1-MANJARO`
- VSCode version: `1.30.1`

:warning: Make sure to **attach** this file from your *home*-directory: `/home/emanuele/platformio.platformio-ide-unresponsive.cpuprofile.txt` :warning:

Find more details here: https://github.com/Microsoft/vscode/wiki/Explain:-extension-causes-high-cpu-load

scopelemanuele, 2019-01-12:
I forgot the file...
[platformio.platformio-ide-unresponsive.cpuprofile.txt](https://github.com/platformio/platformio-vscode-ide/files/2752534/platformio.platformio-ide-unresponsive.cpuprofile.txt)

ivankravets, 2019-01-30:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/478

closed 2019-01-30 by ivankravets