platformio/platformio-vscode-ide #566: File selector widget crashes when non utf8 file encountered

mrbig, 2019-01-26:
Dear Platformio developers!

When I try to navigate into a folder that contains a non utf8 encoded file, then the widget locks at the "Loading..." animation, and there's no way to enter that directory.
See screenshot below:
[image]

I've attached an example bash script that will create such a file for testing purposes.
[test.zip](https://github.com/platformio/platformio-vscode-ide/files/2798915/test.zip)

Expected result: the directory listing should be available even if the file name is not displayed correctly (or not displayed at all)

Thank you

ivankravets, 2019-01-30:
[image]

What is your OS and locale?

mrbig, 2019-01-31:
dear @ivankravets 

Sorry, I forgot about that: I'm running Debian linux (testing). Locale is en_US.UTF8, LC_CTYPE is hu_HU.UTF-8. Filesystem is ext3
Since I have a long time history with linux there are  a bunch of files that were named using hu_HU.iso88592 locale.

mrbig, 2019-02-07:
Note: from the screenshot it looks to me that you're using a C, or iso88591 locale. Since there is not utf8 character handling I believe that's why it's working for you.

ivankravets, 2019-07-17:
Could you reproduce this issue with the latest 1.8.0 release and PIO Core 4.0?

mrbig, 2019-07-18:
@ivankravets I've updated to the 1.8.0, and PIO Core 4.0, also updated vscode to 1.36.1 but the issue still persists.

closed 2020-11-17 by stale[bot]