platformio/platformio-vscode-ide #569: Start PIO Home Server

area81, 2019-01-29:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.2
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 9, in <module>
  File "C:\Users\user1\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <module>
    from twisted.internet._win32serialport import SerialPort
  File "C:\Users\user1\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, in <module>
    import win32file, win32event
ImportError: DLL load failed: ���� ���� ������������ ���������� ���������� ����� BitLocker. ��������� � ������ ����������, ����� �������������� ����.

	at r.P.Promise (C:\Users\user1\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\user1\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\user1\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.__dirname.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

area81, 2019-01-31:
Ivan, BitLocker it's a windows 10 disk encryption software. where is the white list?

ivankravets, 2019-01-31:
Do you have enabled network firewall?

area81, 2019-02-01:
да
yes

> Do you have enabled network firewall?

ivankravets, 2019-02-02:
Could you try to disable it for a while and restart IDE? Just for a test. Thanks!

closed 2019-02-22 by ivankravets