platformio/platformio-vscode-ide #571: Installation Manager

ikwic, 2019-01-30:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.2
PIO IDE: v1.6.0
System: Windows_NT, 6.1.7601, x64

# Exception
```
TypeError: Path must be a string. Received undefined
	at assertPath (path.js:28:11)
	at Object.basename (path.js:834:5)
	at p.some.r (D:\Users\semendyaev\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5321)
	at Array.some (<anonymous>)
	at Object.d [as runCommand] (D:\Users\semendyaev\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5302)
	at f.value.Promise (D:\Users\semendyaev\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:26053)
	at new Promise (<anonymous>)
	at t.value (D:\Users\semendyaev\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:26031)
	at <anonymous>
	at process._tickCallback (internal/process/next_tick.js:188:7)
```

ikwic, 2019-02-01:
Today the problem is gone. Maybe it was caused by unstable proxy server connection.

ivankravets, 2019-02-02:
Thanks for the info! Happy coding with PlatformIO!

closed 2019-02-02 by ivankravets