platformio/platformio-vscode-ide #575: Start PIO Home Server

Raukores92, 2019-02-02:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.30.2
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17763, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 37, in <module>
Exception: PIO Plus: Runtime error

	at r.P.Promise (C:\Users\Adrián\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\Adrián\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\Users\Adrián\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.__dirname.ChildProcess._handle.onexit (internal/child_process.js:209:5)
```

gnikeht, 2019-02-17:
Same here... Cannot start Platform IDE

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 2, in <module>
  File "/Users/gnikeht/.platformio/packages/contrib-pysite/twisted/internet/threads.py", line 19, in <module>
    from twisted.internet import defer
  File "/Users/gnikeht/.platformio/packages/contrib-pysite/twisted/internet/defer.py", line 21, in <module>
    import attr
ImportError: No module named attr

	at P.Promise (/Users/gnikeht/.vscode/extensions/platformio.platformio-ide-1.6.0/node_modules/platformio-node-helpers/dist/index.js:1:14992)
	at ChildProcess.h (/Users/gnikeht/.vscode/extensions/platformio.platformio-ide-1.6.0/node_modules/platformio-node-helpers/dist/index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

ivankravets, 2019-02-22:
@gnikeht  please remove `/Users/gnikeht/.platformio/packages/contrib-pysite` folder and restart VSCode.

closed 2019-02-22 by ivankravets