platformio/platformio-vscode-ide #588: Start PIO Home Server

zoltanszalontay, 2019-02-14:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.31.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 9, in <module>
  File "C:\Users\szaz\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <module>
    from twisted.internet._win32serialport import SerialPort
  File "C:\Users\szaz\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, in <module>
    import win32file, win32event
ImportError: DLL load failed: A megadott modul nem tal�lhat�.

    at P.Promise (C:\Users\szaz\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:177:28)
    at ChildProcess.h (C:\Users\szaz\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\misc.js:167:5)
    at ChildProcess.emit (events.js:182:13)
    at ChildProcess.cp.emit (C:\Users\szaz\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:961:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

zoltanszalontay, 2019-02-14:
Anaconda env for Python 2.7 activated. PIO extension installed. PYTHONPATH is set to Python 2.7 interpreter. PIO Home does not launch with errmsg above.

samoooop, 2019-02-16:
#586

zoltanszalontay, 2019-02-16:
Problem solved. Thanks a lot.

closed 2019-02-16 by zoltanszalontay