platformio/platformio-vscode-ide #601: Start PIO Home Server

jesp1999, 2019-02-20:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.31.1
PIO IDE: v1.6.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 38, in <module>
  File "<string>", line 18, in load_module
  File "<string>", line 9, in <module>
  File "C:\Users\Julian\.platformio\packages\contrib-pysite\twisted\internet\serialport.py", line 87, in <module>
    from twisted.internet._win32serialport import SerialPort
  File "C:\Users\Julian\.platformio\packages\contrib-pysite\twisted\internet\_win32serialport.py", line 19, in <module>
    import win32file, win32event
ImportError: DLL load failed: The specified module could not be found.

	at P.Promise (C:\Users\Julian\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:14992)
	at ChildProcess.h (C:\Users\Julian\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\platformio-node-helpers\dist\index.js:1:5916)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (C:\Users\Julian\.vscode\extensions\platformio.platformio-ide-1.6.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:961:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
```

vitormhenrique, 2019-02-21:
was getting the same issue this fixed for me https://github.com/platformio/platformio-vscode-ide/issues/586

ivankravets, 2019-02-22:
Duplicated of https://github.com/platformio/platformio-vscode-ide/issues/586

closed 2019-02-22 by ivankravets