platformio/platformio-vscode-ide #603: Arduino.h: No such file or directory

leobel96, 2019-02-20:
Hello. I've the last version of PlatformIO for VScode and I've noted an issue. If I create a new project for a board (for example Digispark USB) and then I compile it everything is ok. If I change the board in "platformio.ini" to something different (for example Arduino UNO) I obtain the following error:
```

src\main.cpp:1:21: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://platformio.org/lib/search?query=header:Arduino.h
*
*****************************************************************

compilation terminated.
*** [.pioenvs\uno\src\main.o] Error 1
```
It seems like Platformio doesn't find Arduino.h library.

ivankravets, 2019-02-22:
Please remove `%USERHOME%/.platformio` folder and start build again.

leobel96, 2019-02-22:
I've already done it but nothing changed

ivankravets, 2019-02-22:
Please share your `platformio.ini` here.

leobel96, 2019-02-22:
I've understood the problem: I've always to explicit `framework = arduino` even if it is the only framework a board supports.

ivankravets, 2019-02-22:
Can I close this issue?

leobel96, 2019-02-22:
Yes but is it normal that I've always to explicit it even if a board supports only that framework?

ivankravets, 2019-02-22:
If you plan to use Arduino API, `framework = arduino` is required. Otherwise, you will need  to use native API or Assembly.

closed 2019-02-22 by ivankravets